imxrt-rs/imxrt-hal #5: Upload to EVK

alexandruradovici, 2020-02-20:
Hi guys,

Great work! A quick question. Did you succeed in loading this to an imxrt1052 EVK device? I am trying to upload the following program using mcuxpresso's GUI Flash tool, it seems to upload it but fails to light up the LED.

I would appreciate any feedback :)

```rust
//! Prints "Hello, world!" on the host console using semihosting

#![no_main]
#![no_std]

extern crate panic_halt;

use cortex_m_rt::entry;
use cortex_m_semihosting::{debug, hprintln};
use imxrt_ral::{read_reg, write_reg, modify_reg, reset_reg};
use imxrt_ral::{gpio};

fn set_pin_9(gpio: &gpio::RegisterBlock) {
    write_reg!(gpio, gpio, GDIR, 1<<9);
    write_reg!(gpio, gpio, DR, 1<<9);
}

#[entry]
fn main() -> ! {
    // exit QEMU
    // NOTE do not run this on hardware; it can corrupt OpenOCD state
    // debug::exit(debug::EXIT_SUCCESS);

    let gpio1 = gpio::GPIO1::take().unwrap();
    set_pin_9 (&gpio1);
    loop {}
}
````

memory.x
````
MEMORY
{
  /* Define each memory region */
  FLASH (rx) : ORIGIN = 0x60000000, LENGTH = 0x4000000 /* 64M bytes (alias Flash) */  
  RAM (rwx) : ORIGIN = 0x20000000, LENGTH = 0x20000 /* 128K bytes (alias RAM) */  
}
```

bfrog, 2020-02-20:
You most likely need to setup the boot-gen stuff as well. I've been working on the GPIO side of this to get it going on my 1062 evk and 1011 evk. I'm hoping to try that stuff out this week and get a sample app going. If you beat me to it, please do share so we can put it up here!

bfrog, 2020-02-21:
So some updates on this, I did try to build and run a sample on the board in a few different ways, and when it didn't run right away I started digging. It looks like perhaps some linker script issues that I need to sort out. I've made some issues in this repo and the imxrt-rt repo if you'd like to follow along or add to the discussions.

alexandruradovici, 2020-02-22:
I seems that the elf binary requires a .boot_hdr section (that contains the interrupt vector table) and has the FSCB (FlexSPI Configuration Block). I am still trying to figure out how to set up this. I will keep you posted.

bfrog, 2020-02-22:
So looking closer at the NXP SDK they enable some clocks prior to things working.

I've managed to get a pretty similar example working on my 1060evk today finally, after fighting some linker issues. imxrt-rt *needs* to use the gnu linker not the default llvm linker for the provided link.x script it provides. I'll look into maybe fixing that in the future.

bfrog, 2020-02-26:
So final conclusion here is that you'd likely need to enable some clocks and set the iomux's pin config to be ALT5 for gpio usage.

The hal will take care of all of this as we move forward. Let me know if that helps or if there's more we can do to help in the meantime.

bfrog, 2020-04-05:
Just an update, the 1060evk now has blinking leds, uarts, timers, and pwm working. I'm going to close this issue unless you intend on working on a 1050evk BSP and feature flag for the hal, which we would gladly accept!

bfrog, 2020-04-08:
Feel free to reopen or open new issues if you like. Closing for now

closed 2020-04-08 by bfrog