imxrt-rs/imxrt-hal #8: Port HAL to the RAL

mciantyre, 2020-02-26:
The issue tracks how much of the HAL is migrated from the PAC-based implementation to the RAL-based implementation.

- [x] CCM
- [ ] GPIO
- [ ] I2C
- [ ] IOMUXC
- [ ] PIT
- [ ] PWM
- [ ] UART

My focus is on keeping parity with the existing features; for example, I don't intend on generalizing the HAL for other iMXRTs. Once I can build the HAL, I can show equivalence by [using the examples in the `teensy4-examples` crate](https://github.com/mciantyre/teensy4-rs/tree/master/teensy4-examples/src).

mciantyre, 2020-02-26:
IOMUXC is WIP on `hal_iomux`

mciantyre, 2020-02-27:
Ported IOMUXC in #10

bfrog, 2020-02-27:
Initial iomuxc work done and merged, on to the next thing. I plan on trying to deal with GPIO stuff this week so I can get the typical blinky led example working on the 1060evk. I also picked up a teensy so I can start testing the teensy examples you've made.

mciantyre, 2020-02-27:
Awesome thanks. I can start the I2C peripheral unless there’s something else you’d like to see.

bfrog, 2020-02-27:
I2C sounds great, will be super happy when we get a uart going as well :+1:

mciantyre, 2020-02-29:
I2C port proposed in #11.

bfrog, 2020-03-04:
GPIO in progress, but noticed I think a case where our unsafe usage really is in fact unsafe, see #21. Something I think we'll need to consider what each option means

bfrog, 2020-03-09:
GPIO wrapped up, though future improvements to iomux+gpio are definitely possible to better take advantage of rusts ownership and mutability rules to avoid the critical section

bfrog, 2020-03-09:
@mciantyre did you want to try to tackle pwm or should I? I can hammer it out here I think pretty quick, don't want to step on your toes if you are already on it

mciantyre, 2020-03-11:
I haven't started on PWM yet, but I'm happy to pick it up. I might not get to it until this weekend.

PWM might be the least straightforward to port. When we implemented the driver on the `svd2rust` output, we were taking advantage of SVD `<cluster>`s being represented as a struct. But, the RAL approach doesn't represent SVD clusters (#2); it simply duplicates the registers. This might require us to change a good bit of the implementation; it might be more macro-heavy. There might be some parts of the RAL API that I'm not considering, so I'm going to study up on that.

If it makes it easier to port the PWM driver, I might consider implementing [`embedded_hal::Pwm`](https://docs.rs/embedded-hal/0.2.3/embedded_hal/trait.Pwm.html) instead of [`embedded_hal::PwmPin`](https://docs.rs/embedded-hal/0.2.3/embedded_hal/trait.PwmPin.html).

bfrog, 2020-03-12:
I'll let you tackle PWM then, it sounds like you have some good ideas on what needs to happen.

I'm going to start looking at making sure what we have works on my 1060 evk and fixing bugs as I find them as time permits.

 I can help with the teensy testing when that time comes.

Once things are mostly working on the 1060evk and teensy 4 I'm happy to say this is 0.1 worthy and iterate from there if you are. I certainly have some ideas on where to go next we can maybe chat about in issues or on a video meetup/voice meet if that's something you are interested in at all.

bfrog, 2020-04-02:
I think we can safely close this issue, unless you know of anything left to port @mciantyre

mciantyre, 2020-04-02:
All good!

closed 2020-04-02 by mciantyre