imxrt-rs/imxrt-hal #18: USB Support

bfrog, 2020-03-03:
I'd really like to support USB for audio/midi/mass storage purposes.

mciantyre, 2020-03-05:
I played around with a [`UsbBus` implementation](https://docs.rs/usb-device/0.2.5/usb_device/bus/trait.UsbBus.html) in the `teensy4-rs` repo, but didn't get anywhere notable (mciantyre/teensy4-rs#42). Found [this application note](https://www.nxp.com/docs/en/application-note/AN3631.pdf) useful for understanding the data structures, although that document isn't specific to our processor family. Ideas seemed consistent with what I saw in our processor's reference manual.

Would it make sense to have this feature in a separate repo? I noticed that's how the [`stm32-rs` project does it](https://github.com/stm32-rs/stm32-usbd). But, it looks like they vendored the PAC / RAL modules into the repo. It could be nice to have it in here, so we can directly depend on the RAL.

bfrog, 2020-03-05:
I think I'd rather support all direct peripherals with the hal to start myself.

bfrog, 2020-04-05:
One thing I realized in looking into UsbBus is it doesn't yet support isochronous endpoints (audio) which is a bit disappointing, though perhaps workable still. Something to look into more after we get some basics (HID, virtual console) working.

leo60228, 2020-04-29:
Has there been any more work on this?

mciantyre, 2020-04-29:
No updates on my end.

I'm still using the Teensy 4's standard USB stack, written in C, in the `teensy4-rs` project.

bfrog, 2020-05-18:
None yet from me either, I'm waiting on isochronous support in the stack before really digging in, as the main use case I want is audio. I'd welcome any PRs getting us started before then!

bfrog, 2020-09-03:
I'm going to get the usb-device interface working as a starting point, then see where that takes me

bfrog, 2020-10-06:
There's a branch with some basic device structure setup, it's a fill in the blank project of pll/power/irq handling

noahbkim, 2020-12-03:
Hey all, I'm fairly new to Rust and I know this is probably an unreasonable undertaking for me, but I really wanted to try working on keyboard firmware in Rust with a teensy 4 and eventually landed here as a starting point. I may not be the best candidate, but I would love to contribute to rewriting the USB driver in Rust.

I started putting together a crate with the existing code in the `imxrt-usb` branch, and the first actual issue I ran into is that the proposed `Device<M>` struct is not threadsafe: 

```
error[E0277]: `*const imxrt_ral::usb::RegisterBlock` cannot be shared between threads safely
  --> src/device.rs:31:9
   |
31 | impl<M> UsbBus for Device<M>
   |         ^^^^^^ `*const imxrt_ral::usb::RegisterBlock` cannot be shared between threads safely
   | 
  ::: /Users/noahbkim/.cargo/registry/src/github.com-1ecc6299db9ec823/usb-device-0.2.7/src/bus.rs:17:19
   |
17 | pub trait UsbBus: Sync + Sized {
   |                   ---- required by this bound in `usb_device::bus::UsbBus`
   |
   = help: within `device::Device<M>`, the trait `core::marker::Sync` is not implemented for `*const imxrt_ral::usb::RegisterBlock`
   = note: required because it appears within the type `core::marker::PhantomData<*const imxrt_ral::usb::RegisterBlock>`
   = note: required because it appears within the type `imxrt_ral::usb::Instance`
   = note: required because it appears within the type `device::Device<M>`
```

In general, I cannot see how register pointers could ever be used in a threadsafe capacity. Could someone explain what I'm missing here/any suggestions of where to start looking/any references I should read through before continuing to tinker with this stuff?

noahbkim, 2020-12-03:
Additionally, if I were to use the [`stm32-usbd`](https://github.com/stm32-rs/stm32-usbd) implementation as reference, what major differences do you think there might be between hardwares?

bfrog, 2020-12-04:
You can definitely refer to the stm32-usbd as a reference point. The stm32-rs project is mostly built around svd2rust generated register mappings, which for imxrt really didn't make sense to use (it took 10+ minutes to build the 1mloc+ svd2rust generated).

We instead based our register mapping on stm32ral, written by the stm32-rs project lead.

Point is there will be a difference in the way registers are accessed. You can always create your own register mapping as well if it makes sense. It might! You should keep in mind that there is the intent to support more than imxrt1062.

The sync issue above is likely being caused by the const *. You can instead take the Instance type, which your peripheral driver then owns and there shouldn't be any Sync issues.

mciantyre, 2020-12-05:
> what major differences do you think there might be between hardwares?

I believe the STM32 USB peripherals only support full-speed transfers. The i.MX RT USB peripherals support high-speed transfers. Since our peripheral needs to support faster transfers, there will be differences in the peripheral designs, particularly around transfer scheduling.

I'll recommend [this application note](https://www.nxp.com/docs/en/application-note/AN3631.pdf) to get a sense of how the i.MX RT USB peripheral works. The AN is for a different processor, but the i.MX RT chips use the same names, data structures, and patterns. I found this AN easier to read than the reference manual.

> I cannot see how register pointers could ever be used in a threadsafe capacity.

Your intuition is right. It's our job to make sure its `Sync`. We'll need either a [`Mutex`](https://docs.rs/cortex-m/0.7.0/cortex_m/interrupt/struct.Mutex.html), or well-considered `unsafe` code.

noahbkim, 2020-12-05:
Thanks for all the pointers, @bfrog @mciantyre! I'm not sure how much I'll be able help, but I'll definitely give it a shot. If you'd like me to put followup questions in another thread to keep this one more focused on actual development progress, I can open another issue.

mciantyre, 2021-02-28:
I put together a proof-of-concept driver in the [imxrt-usbd](https://github.com/imxrt-rs/imxrt-usbd) repo. The full-speed driver implements the `UsbBus` trait from the `usb-device` crate. There's an example that shows a USB CDC device for a Teensy 4.

mciantyre, 2021-03-11:
There's a 0.1 release on crates.io. Bring future contributions to [the dedicated repo](https://github.com/imxrt-rs/imxrt-usbd).

closed 2021-03-11 by mciantyre