imxrt-rs/imxrt-hal #19: SAI Support

bfrog, 2020-03-03:
SAI support would be a great way to make good use of these chips as audio processors

bfrog, 2020-05-18:
I'm waiting until we have a solid DMA branch before really delving into this. DMA is something I'd like to immediately support with the driver.

In the meantime I'm working on a little no_std modular synth/sampler/sequencer type library for the project I have in mind.

bfrog, 2020-08-03:
I think I might be waiting forever on the audio support in usb-device unless I start looking to do it myself, which is what I was looking to see before committing time on this

ChristianS99, 2021-07-28:
I want to use a Teensy4.1 with the audio adapter, that is attached via i2s. what needs to be done for that to work?
You mentioned proper DMA for this, what is the state of DMA? Does it need to be implemented in this create, or somewhere else?

BR, Christian

bfrog, 2021-08-03:
You can do it without DMA, the chances of missing the fill for the FIFO and getting an underflow or overflow error I think increases though as now you need to constantly attempt to keep the SAI FIFO filled (for tx) and drained (for rx).

mciantyre, 2021-08-07:
> I want to use a Teensy4.1 with the audio adapter, that is attached via i2s. what needs to be done for that to work?

imxrt-rs/imxrt-iomuxc#3 added SAI pins, so we're set on hardware pin support. To support an I2S adapter on the Teensy 4.1, we would need an SAI driver in the HAL.

> what is the state of DMA? Does it need to be implemented in this create, or somewhere else?

We have a proof-of-concept [DMA driver](https://github.com/imxrt-rs/imxrt-hal/blob/cd6e3f50c4f6d513047cec149007b4eac4f1f9a3/imxrt-hal/src/dma.rs#L1) in the 0.4 HAL. There's DMA support for SPI, UART, and ADC. SAI support might look similar; just implement [a few traits](https://github.com/imxrt-rs/imxrt-hal/blob/cd6e3f50c4f6d513047cec149007b4eac4f1f9a3/imxrt-hal/src/dma/peripheral.rs#L4).

prutschman, 2022-06-15:
I have just started experimenting a bit with this. I agree that DMA is the better way to go here, but if you trigger an interrupt based on the FIFO watermark, and aim to write 4-8 samples per interrupt, the CPU overhead is quite manageable.

teburd, 2022-06-15:
That of course makes perfect sense, very cool