imxrt-rs/imxrt-hal #30: Default pin mux option wrong for evk

bfrog, 2020-03-24:
The teensy appears to do some work in its bootloader to setup a few things, and avoid the setup work normally needed on imxrt chips.

In order to get my led blinking (finally did!)

I needed to change the default alt of the pads to alt0 as that is in fact the default. Teensy must be setting these up to Alt5 by default. This needs to be reconciled.

I also needed to disable a few watchdogs otherwise the chip would reset very quickly leaving the debugger and myself very confused.

Lastly the runtime for the teensy makes some assumptions that I don't think are true for the evk boards.

All in all I think we need to look at how the HAL can provide the tools, but not necessarily supply the default states assumed to be there currently to account for the variations between the teensy environment and the bare imxrt1062 environment found on the EVK boards.

See the imxrt1060evk_fixes branch for everything I needed to change to make this work.

mciantyre, 2020-03-25:
Glad to hear you got output on the EVK board!

> I needed to change the default alt of the pads to alt0 as that is in fact the default.

The IMXRT1060 reference manual, section 11.7, states that the reset value for nearly all of the `SW_MUX_CTL_PAD_GPIO_*` registers is `0x5`, or `Alt5`. We also see this in the SVD files, which [propagate to the reset values for the IOMUXC registers](https://github.com/imxrt-rs/imxrt-rs/blob/87e5408f1d60f1561abeec120c8fc2f6fc949072/imxrt-ral/src/imxrt106/instances/iomuxc.rs#L193) (exposition only; RAL is no longer checked in). There's no additional setup in the Teensy runtime crate to set these values across the MUX registers. We've selected `Alt5` as the default type state because it reflects the reset value.

From a scan of the `SW_MUX_CTL_PAD_GPIO_*` table, there are only eight registers that default to `Alt0`:

1. `AD_B0_04`
2. `AD_B0_05`
3. `AD_B0_06`
4. `AD_B0_07`
5. `AD_B0_08`
6. `AD_B0_09`
7. `AD_B0_10`
8. `AD_B0_11`

#31 exposes those eight pads in the IOMUXC. Are those eight pins used on the EVK?

I agree that those eight pads should default to an `Alt0` type state. But, we're changing all of the pad type states from `Alt5` to `Alt0`. Are we sure that's appropriate?

bfrog, 2020-03-25:
Well, that is great news actually! I did not see the defaults listed, something I should have looked at. I'll change the defaults to match the ref manual, this makes a lot of what I said sound silly of course. Thanks Ian. I made an assumption after seeing the register reading back 0x00 that they were all like that. This I think solves the mystery.

closed 2020-03-25 by bfrog