imxrt-rs/imxrt-hal #33: sion_enable/disable safety

bfrog, 2020-03-27:
I noticed when adding the pin configure functionality that sion_enable/disable take &self rather than &mut self, while internally mutating a register, this is possibly unsafe as two references can now mutate the state of the pin leading to confusing situations where perhaps multiple references of the pin ask for different sion enable states. It's unlikely, but it is still unsafe.

mciantyre, 2020-03-28:
Good catch. Can we change it to `&mut self`? It's an internal method, so this shouldn't affect anyone. Would that give us the confidence that we're safe?

bfrog, 2020-03-28:
I believe &mut self is perfectly fine here. Will change it here tomorrow when I get a free moment hopefully and, crossing my fingers, it doesn't break the compile

commit ac2d309c (2020-04-03): Ensures sion_enabled/sion_disable safety

Fixes #33
PR #39 (2020-04-03): Ensures sion_enabled/sion_disable safety
Fixes #33
closed 2020-04-03 by bfrog