imxrt-rs/imxrt-hal #41: Release 0.2.0

bfrog, 2020-04-05:
I'm very satisfied with where things are I think. Enough so I think its time to get the next minor pre 1.0 revision out there in the wild.

I've got most of the basic things working on an imxrt1060evk at this point.

I've tested some of the examples on the teensy and was happy enough with the results, I don't have a MPU9250 to test the i2c/spi examples with.

If there's anything left todo before then please note that here. Will wait til Apr 8 before tagging and bagging.

bfrog, 2020-04-05:
As noted, remove setting the default run mode and document the need to set it to use SysTick to interrupt sleep.

mciantyre, 2020-04-07:
All good here! I reran my Teensy 4 tests on e51c5aa4f652f62c7963c9ac7b32fde7b2246678. Thanks for coordinating the 0.2.0 release.

bfrog, 2020-04-08:
0.2.0 is out

closed 2020-04-08 by bfrog