imxrt-rs/imxrt-hal #43: Support SPI DMA transfers

mciantyre, 2020-04-07:
From @dflemstr:

> I love the recent development and buzz around SPI support. In my application, I want to transfer a lot of data, and it would hence be useful to support DMA transfers for SPI!
>
> I have another side project where I've been trying to eek out the maximum amount of performance from SPI in a device independent manner. I defined the API to look like this: https://github.com/dflemstr/embedded-platform/blob/master/src/spi.rs
> 
> Of course, that introduces async as well, while this crate/embedded_hal is very focused on blocking APIs. However the main take-away from that API I think is:
> 
> You initiate a transfer which takes ownership of the slice of memory you want to send
You can poll that transfer to completion (and do that in a tight loop if you want a "blocking" API)

Manually migrating this from mciantyre/teensy4-rs#51

mciantyre, 2020-04-21:
Anyone mind if I take this on? I've been playing around with the DMA controller, and I have a working DMA-based UART driver in another project. My second pass should be better than my first 😛

mciantyre, 2020-04-28:
@dflemstr, if you have the time, I'd love your thoughts on #49. I've demonstrated full-duplex SPI transfers using DMA in [the `teensy4-rs` repo](https://github.com/mciantyre/teensy4-rs/blob/0ba381b8c30e04005bb3ae7e50b579c199f97679/teensy4-examples/src/dma_spi.rs), and it'd be good to see if this could satisfy your use-case.

dflemstr, 2020-05-12:
This is looking fantastic! Due to covid and other things I have had to postpone making progress, but I will make an attempt to port over the code for my use-case on top of your feature branch!

PR #49 (2020-05-27): Add DMA channels
The PR adds DMA channels, and a set of traits that lets us add DMA capabilities for iMXRT peripherals. I extend the UART peripheral with the new DMA capabilities.

There's [a rough example in the `teensy4-rs` repo](https://github.com/mciantyre/teensy4-rs/blob/feature/dma/teensy4-examples/src/dma_uart.rs) showing how we can use DMA to coordinate UART transfers and receives. It needs to be cleaned up and documented, but it seems to work!

There are a bunch of DMA capabilities that this PR does not support, like

- channel arbitration modes. The DMA controller supports fixed-prio [... cut]
closed 2020-05-27 by mciantyre