imxrt-rs/imxrt-hal #44: Cannot build 0.2.0 crates from crates.io

mciantyre, 2020-04-09:
I'm hitting build errors when referencing the 0.2.0 releases from crates.io. It affects the RAL, and maybe the HAL.

Minimal reproduction of the RAL:

```
cd /tmp
cargo new --lib test-ral
cd test-ral
echo 'imxrt-ral = { "version" = "0.2.0", features = ["imxrt1062"] }' >> Cargo.toml
cargo build

...

error: couldn't read /Users/mciantyre/.cargo/registry/src/github.com-1ecc6299db9ec823/imxrt-ral-0.2.0/src: Is a directory (os error 21)
```

It looks like 742e687 explicitly sets the RAL library's path to `src`. Does that need to be the path to the library entrypoint, `src/lib.rs`? Is there a reason we're explicitly setting the library path?

Additionally, the [HAL references a local directory containing a 0.2.0 release of the RAL](https://github.com/imxrt-rs/imxrt-rs/blob/742e68757abc9d1a0595ab3105c24e9aa8f27497/imxrt-hal/Cargo.toml#L17). Should the path be there? I'm actually not sure if that will be troublesome for end users.

mciantyre, 2020-04-09:
I'm playing around with a `0.2.1-alpha` release of both crates. Goal is to download the HAL from crates.io in the `teensy4-rs` project.

mciantyre, 2020-04-09:
Wanted to see if this was system-dependent, and it seems that it is! I cannot reproduce this in a Linux VM.

mciantyre, 2020-04-09:
We may want to consider yanking the 0.2.0 release. As of this writing, I've not yanked it. I'd like for someone else to reproduce the issue before we consider a yank.

mciantyre, 2020-04-09:
I've release 0.2.1 of both the RAL and HAL to crates.io. See #45.

bfrog, 2020-04-09:
Thanks! I added that src because cargo publish was complaining about not being able to find src/lib.rs, not sure why. I fixed that, only to run into other issues, said forget it and can publish --no-verify, thanks for fixing this!

bfrog, 2020-04-10:
We should probably yank 0.2.0

closed 2020-04-10 by bfrog