imxrt-rs/imxrt-hal #47: ADC implementation

Walther, 2020-04-13:
`embedded-hal` specifies ADC traits [here](https://docs.rs/embedded-hal/0.2.3/embedded_hal/adc/index.html)

It seems that this crate does not implement those yet. This would be useful in order to e.g. read analog values from the pins of a Teensy 4 board.

---

_Would love to try and help with this, if someone could point me to the right direction & provide a bit of mentoring 🙂_

bfrog, 2020-04-13:
Hi Walther!

ADC would be a great addition to the HAL.  In looking at the embedded-hal interface its pretty limited in how it allows the ADC to operate. Still its a good start for a one shot software drive conversion I suppose! Hopefully they have improvements in mind. 

I highly recommend keeping a copy of the enormous ref manual handy. You might need to signup for a NXP account to download it.

To implement the Channel trait I might define a macro for such a thing in a new adc.rs

```
macro_rules! adc_channel {
   ($adc: ident, $channel: ident, $pin: ident, $alt_mode: ident) => {
        impl Channel<$adc_instance> for $pin<$alt> {
        }
   }
}
```

Implementing OneShot I'd likely do something similar, though it'd require perhaps some thought on how the one shot works, we likely can't just impl OneShot<SomeGPIO<Alt5>> { ... } safely

Because even though it takes a mutable reference to the GPIO it does not necessarily provide safety around some of the shared ADC registers. You can solve the unsafety a few different ways but probably the sanest is to have an ADC take ownership of the pins you'd like to use ADC with and have that new ADC type then provide access to each of the ADC inputs as a embedded_hal OneShot. I'd look at how the PWM peripheral in the HAL is doing something similar.

The ref manual describes what must be done to do a one shot software triggered conversion at 66.5.4 If examples are needed I'd look at the mcuxpresso sdk and the examples provided by it to get an idea of what all they do to enable conversions.

DavidTheFighter, 2020-09-27:
Any updates on this? I'd be happy to try and get something working but I'm new to both embedded and rust so it might be an interesting time.

mciantyre, 2020-09-28:
Haven't seen any updates on my end. Happy to help with either the embedded or the Rust parts.

The [CONTRIBUTING](https://github.com/imxrt-rs/imxrt-rs/blob/master/CONTRIBUTING.md) guide gives some tips to get started, and points to resources you might want to have on hand, like the processor reference manual.

bfrog, 2020-09-29:
I might avoid the macro at this point, I don't know if its necessary with the way we are doing things now. Instead use generics if possible, as they are easier to document.

Walther, 2020-09-29:
Sorry for the lack of updates!
I ended up thoroughly confused by the significant use of macros in this crate, not having personally implemented / used any before. I would need to learn writing those first with a simpler project / tutorial, read more about them in the book, etc.

I still have the devices and have some audio projects in mind, so I would be happy to help with testing if you do get something implemented!

DavidTheFighter, 2020-10-02:
Got some bare basic test code working, I'm thinking about how to do the interface. First, there's a LOT of register settings (ADHSC, ADLPC, ADLSMP, ADSTS, and the clock dividing vs asynchronous clock, see pg. 3354 in the RT1060 manual). Do we want to expose each of these settings individually or do more what like the teensy core library and others do and hide those settings behind the user's choice for resolution and/or sampling/conversion speed. Or both? The iomuxc crate is also a bit of a mystery for me, and I'm not sure how to get channels from pins (eg. pad AD_B1_02 is channel 7, for the input select of the ADCx_HC0 register).

mciantyre, 2020-10-03:
> Do we want to expose each of these settings individually or do more what like the teensy core library and others do and hide those settings behind the user's choice for resolution and/or sampling/conversion speed.

I'll vote for the simpler API that lets users configure the driver by resolutions and conversion speeds. An API using resolutions might be easier to use for newcomers to this chip. If we find a need for a power-user API in the future, we should be able to expose those methods in a forward-compatible manner. If you'd like to tackle both as part of this effort, go for it!

Some other HAL peripheral drivers let users select clocks (kinda; most clock selections are only a single option). So, if you're thinking about letting users select the IPG clock, IPG div 2 clock, or ADACK, there's examples for how we might do it. That being said, the ADC driver implementation might be simpler by committing to just ADACK. The [data sheets](https://www.nxp.com/products/processors-and-microcontrollers/arm-microcontrollers/i-mx-rt-crossover-mcus:IMX-RT-SERIES) note ADACK clock speeds; I couldn't find it in the reference manual.

> The iomuxc crate is also a bit of a mystery for me, and I'm not sure how to get channels from pins

I might be able to get you started. The goal of the IOMUXC crate is to specify which pads support which functions. There are tables in the reference manual (like Table 10-1, Table 66-2, iMXRT1060, Rev 2) that indicate the ADC-compatible pins. We just need a way to express that in Rust.

[This commit](https://github.com/mciantyre/imxrt-rs/commit/06396b12277ead82233f5be16775d26af01a372c) in my `imxrt-rs` fork, [branch `proto/adc`](https://github.com/mciantyre/imxrt-rs/tree/proto/adc), prototypes one way to express the ADC-compatible pins. It should handle the fact that some pads might be used for both ADC1 and ADC2. But, it's otherwise untested, and has a few TODOs. Feel free to steal it, ignore it, or ask questions right on that commit.

From that IOMUXC implementation, the pads express their supported ADC module:

```rust
use imxrt_hal::iomuxc::adc::{ADC, Pin}

fn use_adc_pin_for_something<ADCx: ADC, P: Pin<ADCx>>(pin: P) {
    // ADCx is either ADC1, or ADC2
    // The INx number is available using the associated type, P::Input:
   let input_channel: usize = P::Input::USIZE;
}
```

DavidTheFighter, 2020-10-04:
Ahh ok that makes sense, thank you! I'll see if I can get a prototype up for you guys to pick over.

PR #84 (2020-10-15): ADC Implementation
This PR adds an ADC functionality and implements the `embedded-hal` 0.2.3 traits (0.2.4 makes a few small changes). It closes #47. 
This has all the features that I will reasonably use, and has a couple extra features for power users. I decided to go with a simpler interface rather than letting the user change every setting possible, so real power users might want more functionality but we can always add that later. This was tested on a Teensy 4.1
Feel free to nitpick over things :)
closed 2020-10-15 by mciantyre