imxrt-rs/imxrt-hal #48: Could you add some examples

RockySong, 2020-04-14:
I'm an engineer in NXP MCU system engineering team, I'm now learning Rust and I'd like to have a try and then contribute. 
I'm familiar with the i.MX RT devices and using Segger J-Link GDB server to debug it with GDB.
Could you just add some examples to let traditonal MCU users like me get started?
You can also contact me with rocky.song@nxp.com

bfrog, 2020-04-14:
Hi Rocky!

We have some examples that will be part of the imxt1060evk-bsp when I release that this week sometime. They are mostly working right now, the I2C and SPI devices need further testing. The repo also is currently using paths rather than released crates. It is definitely my intention to add more examples as time permits.

Most of the usage of the imxrt-hal and imxrt-ral crates are done through board packages, at the moment the teensy4-rs project has board and runtime specific crates for the teensy 4 and imxrt1060evk-bsp, located in this github org.

I'm excited to see interest from some folks at NXP!

RockySong, 2020-04-16:
Glad to get your response！At this moment though I still need to learn a lot to do real work, I can give some generic comments:
1. For "provide all the linker and reset handler work needed", I am familiar with them and customization under C and asm language for ARMCC, IAR, and GCC, and I can help if I know how to connect my knowledge to Rust code.
2. It is straightforward to configure the Segger JLink GDB server to initialize the SDRAM controller, just use "-jlinkscriptfile" option to run SDRAM init script (provided by NXP SDK, the file name is "evkmimxrt1060_sdram_init.jlinkscript").

bfrog, 2020-04-16:
For the linker script and reset handler customization the current goal is to allow a person to customize it in a build.rs, Cargo uses that to do custom build time work which you can code in Rust.

To solve the problem I'm working on creating a linker script and reset generator that an application creator may call in a build.rs file in their project, this work is ongoing https://github.com/imxrt-rt/imxrt-rt-gen. This will go alongside the boot configuration generator we have https://github.com/imxrt-rs/imxrt-boot-gen. With the two of those crates together hopefully any elf build and memory configuration should be possible, with perhaps the exception of the signed or signed and encrypted boot image support these chips have.

This is all based on feedback here, https://github.com/imxrt-rs/imxrt-rt/issues/2

There's some links there which lead to some more useful information where the idea stems from originally, which is from @japaric 

It's also somewhat of a loftier goal to avoid having to write a custom runtime crate altogether and instead let the rust embedded cortex-m-rt crate deal with it, since they are looking to solve the same sorts of problems, and could be done in a way more SoCs could benefit. Especially NXP ones with their Harvard architectures.

I did see the sdram init jlink script, I thought it was a very neat way of doing it! I'll certainly give it a try when I get there.

bfrog, 2020-05-04:
I'm going to close this for now as there are examples. If specific docs or examples are being looked for, let me know!

closed 2020-05-04 by bfrog