imxrt-rs/imxrt-hal #59: Release: HAL 0.3.0, RAL 0.2.2

mciantyre, 2020-06-11:
We've accumulated a decent set of features and fixes in the HAL ([CHANGELOG](https://github.com/imxrt-rs/imxrt-rs/blob/d02bcc339c626ff21ae2a5727bbc4515d5963259/imxrt-hal/CHANGELOG.md#unreleased)). There's one more HAL feature pending in #58. And, once #57 is merged, we will have fixed the RAL's documentation. Given the updates, I'd like to cut a release of both crates.

- `imxrt_hal` minor bump to `0.3.0`
- `imxrt_ral` patch bump to `0.2.2`

Release will gate on #57 and #58. I'm shooting for 2020/06/17 as the earliest release date.

Let me know if we'd like more time to prepare!

mciantyre, 2020-06-13:
I might also include #60, which would increment the RAL to `0.3.0`.

bfrog, 2020-06-15:
This sounds good to me, release when ready

PR #62 (2020-06-19): Prepare 0.3.0 release of RAL and HAL
All builds and publish dry-runs look OK at this point. I updated the release steps to better order the process.

I removed the `documentation` keys from both `Cargo.toml` files. Neither key pointed to a valid page on docs.rs, and it sounds like crates.io will [automatically provide the link](https://doc.rust-lang.org/cargo/reference/manifest.html#the-documentation-field) if we don't specify it.

Closes #59.
closed 2020-06-19 by mciantyre