imxrt-rs/imxrt-hal #93: Automated Tasks for Releasing

bfrog, 2020-10-18:
I believe nrf-hal uses xtask and some automation to bump versions and perhaps do releases.

Might be worthwhile taking a peek before we release again to perhaps automate some of the release steps.

Not a high priority, something I noticed when updating our RELEASE.md and thinking on what I saw in nrf-hal

bfrog, 2020-10-25:
I think we can close this as we are working towards a split repo and this is less meaningful

mciantyre, 2020-10-25:
Could we reopen this issue?

#94 is focused on separating foundational, chip-specific drivers by repositories. It does not describe a repository boundary for chip-specific HALs. Release automation would still be useful when considering the release of a common HAL and split HALs, like

- imxrt-common-hal
- imxrt1010-hal
- imxrt1020-hal
- imxrt1050-hal
- imxrt1060-hal
- imxrt1064-hal

which could still be maintained here.

bfrog, 2020-10-25:
This makes perfect sense to me, sorry I missed that bit