imxrt-rs/imxrt-hal #103: Controller Area Network (CAN)

reneherrero, 2020-11-27:
Hi,

Wondering if anyone has ventured into the Controller Area Network (CAN) space?

Thanks,

mciantyre, 2020-11-27:
I've not worked with CAN before, and I haven't heard of anyone working on a Rust-based CAN driver for these processors. I'm happy to support an i.MX RT CAN driver's development, then maintain it here.

Here's a quick summary of CAN support across the i.MX RT processor families. Let me know if we see a discrepancy.

| i.MX RT | CAN support? | CANFD support? |
| ------- | ------------ | -------------- |
| 1010    |     NO       |       NO       |
| 1015    |     NO       |       NO       |
| 1020    |   **YES**    |       NO       |
| 1050    |   **YES**    |       NO       |
| 1060    |   **YES**    |     **YES**    |
| 1064    |   **YES**    |     **YES**    |

We could implement [`embedded-can`](https://crates.io/crates/embedded-can), then transition to the embedded-hal equivalent traits if / when it's merged.

DavidTheFighter, 2020-12-26:
I have a CANFD only implementation written for the Teensy. It's pretty application specific to what I'm doing with it, and ~~a little~~ very messy code-wise, but it could be a decent reference for anyone else implementing their own. You can find it here: https://github.com/DavidTheFighter/teensy4-canfd
I'm planning on rewriting it at some point so it depends only on `imxrt-hal`, however the only real `teensy4-bsp` dependency is for the interrupt handler.

rnd-ash, 2021-05-10:
I too would love this. I plan on using a Teensy 4.0 as a Transmission ECU for [this project](https://github.com/rnd-ash/ultimate_nag52), and want to use Rust rather than C++.

However, my TCM only uses standard 500kbps CAN 1.0 rather than CAN-FD or CAN-2.0.

I might try and get an implementation working some time in the next couple weeks.

myCatPlaysFetch, 2022-10-26:
I have some decent experience with CAN, and specifically with the 1064, but only in C. I'd like experience with Rust. I could maybe be of some help here. If nothing else, I'm open to offering some support?

talhaHavadar, 2022-10-27:
> I have some decent experience with CAN, and specifically with the 1064, but only in C. I'd like experience with Rust. I could maybe be of some help here. If nothing else, I'm open to offering some support?

Here you can find CAN implementation example in Rust. https://github.com/stm32-rs/bxcan