imxrt-rs/imxrt-hal #105: iMXRT1064 support

hatimbt, 2021-03-21:
We should publish the imxrt-hal with support for the IMXRT1064 device if it is ready.

Continuing discussion from [ imxrt-rs imxrt-boot-gen#7](https://github.com/imxrt-rs/imxrt-boot-gen/issues/7).

> The HAL is still figuring out the simplest way to support all of the iMXRT MCUs. Our current thinking is to define an imxrt1064-hal crate, and have one HAL per reference manual. These are all just ideas, and we'd love your feedback and ideas in the HAL's issue tracker.

I'm fairly new to Rust embedded and I'm not yet fully aware of best practices. Having one HAL for each reference manual seems sensible. To have one HAL for each of the items in [this](https://www.nxp.com/products/processors-and-microcontrollers/arm-microcontrollers/i-mx-rt-crossover-mcus:IMX-RT-SERIES) product list would be clear from the users' perspective. Considering NXP has released an even more powerful [IMXRT1170](https://www.nxp.com/docs/en/nxp/data-sheets/IMXRT1170IEC.pdf) series with 5 different variants, and some of them have dual cores. These variants could be chosen using the feature tags.

> I'm curious as to how to interface that on-board flash. I can't easily find any NXP docs that say "on-board flash starts at this address." (Since the 1064 RM notes that all FlexSPI2 pin muxing registers are "reserved," I'm guessing that on-board flash is available via FlexSPI2.)

The on-board iMXRT1064 flash is at 0x70000000 upto 0x703FFFFF. [This](https://www.nxp.com/docs/en/application-note/AN12290.pdf) document describes the required changes when moving from an iMXRT1060 series device.

The onboard flash does use the FlexSPI2. The onboard flash is actual an embedded [Winbond W25Q32JV](https://www.winbond.com/resource-files/w25q32jv%20spi%20revc%2008302016.pdf)

mciantyre, 2021-03-23:
Thanks for the info and documentation pointers.

#106 proposes 1064 support for the 0.4 HAL. If accepted, I can publish the change as a 0.4.3 release, and you could start experimenting with the crates. Would that work for you?

Longer term, we'll need to figure out how 1064 support fits with split HAL maintenance (#56) and / or #92. Given that its so similar to the 1062, it may make sense to simply bundle it with the 1060 HAL family.

mciantyre, 2021-04-12:
Sorry for the delay! The newly-released imxrt-hal crate (0.4.3) adds an `"imxrt1064"` feature. This will let us use a HAL that's targeted for the 1064. If you're using a HAL with the 1062 feature, simply change the feature flag; everything else should work the same.

PR #106 (2021-03-23): Add support for iMXRT1061 and 1064
The PR adds two new features,

- `"imxrt1061"`
- `"imxrt1064"`

for supporting the 1061 and 1064 MCUs. These should be trivial to support, since they're so similar to the 1062.

Notable differences:

- From brief inspection of the data sheet, the 1061 is missing the LCD/CSI/PXP peripherals. We don't have drivers for these peripherals today, so there's no concern. If they were to be added to the 0.4 HAL, we will have the feature flags to selectively enable the drivers.
- 1064 has 4MB on-board flash connected to FlexSPI2. Given the allocation, the IOMUXC marks  [... cut]
closed 2021-04-12 by mciantyre