imxrt-rs/imxrt-hal #112: Input pin HIGH state persists.

GuiAmPm, 2021-11-11:
Hello!

I'm new to embedded Rust and I'm using this library with Teensy-rs and I'm facing some strange behaviour I'm not sure is intended.
During the setup, I prepared the LED (output, pin13) and a input (pin 0).

```
let mut led = {
   let pin = bsp::hal::gpio::GPIO::new(pins.p13);
   pin.set_fast(true);
   pin.output()
};
let input = bsp::hal::gpio::GPIO::new(pins.p0);
```

Input is connected to 3v by a button.
During the loop I check turn on or off the LED depending on Pin 0 state:
```
if input.is_set() {
   led.set();
} else {
   led.clear();
}
```

Problem is, once the button is pressed, the LED never turns off and the is_set() returns `true` even after rebooting the board. Only after removing the power source, the value is reset.
This does not happen if the code is written in C using Teensyduino.

I have found a workaround by turning the PIN into output and clearing it, but I'm not sure if this is what I'm supposed to do.

```
let mut output = input.output();
output.set_fast(true);
output.clear();
input = output.input();
```

mciantyre, 2021-11-12:
Welcome! This isn't intended behavior. Could you share the equivalent Teensyduino code? Or, can you confirm that you're calling `pinMode(0, INPUT);` in the working example?

Let's try configuring a pulldown resistor on pin 0 before creating the GPIO input. I think this should do it:

```rust
    use bsp::hal::iomuxc::{self, Config};

    const PULL_DOWN: Config = Config::zero()
        .set_hysteresis(iomuxc::Hysteresis::Enabled)
        .set_pull_keep(iomuxc::PullKeep::Enabled)
        .set_pull_keep_select(iomuxc::PullKeepSelect::Pull)
        .set_pullupdown(iomuxc::PullUpDown::Pulldown100k);

    iomuxc::configure(&mut pins.p0, PULL_DOWN);
    let input = bsp::hal::gpio::GPIO::new(pins.p0);
```

It sounds like the keeper might be introducing this troublesome behavior. The "keeper" keeps the last-observed value on the pin, and it's enabled at power up. This might explain why the input goes high, and stays high. In the Teensyduino world, the `pinMode()` function disables the keeper.

GuiAmPm, 2021-11-12:
Thanks for checking this.
Is pinMode(0, INPUT) ? Yes, it is.
```void setup() {
  pinMode(ledPin, OUTPUT);
  pinMode(0, INPUT);
}

void loop() {
  int val = digitalRead(0);
  digitalWrite(ledPin, val);
  delay(100);
}```

Using this the pull down configuration you shared works as expected, the LED turns off after the button is released.

mciantyre, 2021-11-13:
Perfect, thanks for confirming. Reach out if you notice any other issues.

Future readers should also see mciantyre/teensy4-rs#59.

closed 2021-11-13 by mciantyre