imxrt-rs/imxrt-hal #119: uSDHC support

AstraLuma, 2022-07-10:
I'm not seeing  code for the uSDHC/SDIO peripheral found in the RT1062.

Add this please?

AstraLuma, 2022-07-10:
I might end up doing this myself, but I'm pretty new to both rust and rust embedded, and I've got other stuff going on right now.

mciantyre, 2022-07-11:
Happy to help answer any questions here.

Related:

- https://github.com/mciantyre/teensy4-rs/issues/102
- https://github.com/imxrt-rs/imxrt-iomuxc/pull/24

mciantyre, 2023-03-15:
Cherry-picking ideas from a downstream issue.

> If you're interested in FAT file systems, take a look into [embedded-sdmmc](https://docs.rs/embedded-sdmmc/0.4.0/embedded_sdmmc/). Maybe there's a way to implement the `BlockDevice` trait for the uSDHC peripheral.
> 
> Also note that embedded-sdmmc already has some kind of SPI support. [Here's](https://docs.rs/embedded-sdmmc/0.4.0/embedded_sdmmc/sdmmc/struct.BlockSpi.html) the `BlockSpi` adapter, which provides a default `BlocKDevice` implementation. That might already play well with imxrt-hal's SPI drivers, so the "SD over SPI" problem might already be solved.

ameliamariecollins, 2023-03-15:
I'm not sure of this is common knowledge, but I stumbled across a recipe to "misuse" the pins of the onboard SD card to allow use of simple 3-wire SPI:

Pin 44 (CLK) → SCK
Pin 45 (CMD) → PICO (SDO from the controller's point of view)
Pin 43 (DAT0) → POCI (SDI from the controller's point of view)
Pin 46 (DAT3) → CS

This worked somehow. In fact it worked at 25MHz over jumpers in a breadboard. Goodness knows how; it doesn't even line up with LPSPI1's ALTs. (I was running CircuitPython on the Teensy4.1 at the time.)

We shall do better, of course. But in a pinch...