imxrt-rs/imxrt-hal #120: Support for FlexIo

Sandvoxel, 2022-09-25:
Is there planned support for access to FlexIo on the imxrt?

Sandvoxel, 2022-09-26:
I have some knowledge of the hardware registers for the imxrt FlexIO registers as I am using them in a library in C++ to access a flash chip that use QSpi. I would be happy to help out with writing the library to access FlexIo but I'm quite new to rust and don't really know to start on direct memory access to set up registers. I saw this [issue ](https://github.com/imxrt-rs/imxrt-hal/issues/117) on the matter but it seems to have gone cold. My current project is a teensy 4.1 custom board that is planed to be used as a flight computer in the future but is in the early stages (Just completed the first dev board). I would like to use rust over C++ but I would need more access to the hardware such as FlexIo for the onboard flash chip for data logging.

Sandvoxel, 2022-09-26:
I see that is this may have been done with the imxrt-ral library. I would still like to hear input on the matter.

mciantyre, 2022-09-27:
I'm personally not planning FlexIO support right now. I'm happy to support you or anyone who wants to tackle it.

I recommend starting with [`imxrt-iomuxc`](https://github.com/imxrt-rs/imxrt-iomuxc). Add a marker trait for FlexIO-compatible pins, and implement it for the Teensy 4.1 pins that you need. A few contributors have gone through the process of adding new traits and pins, so check out the merged PRs to see the process.

> I see that is this may have been done with the imxrt-ral library.

Yup, I recommend using [`imxrt-ral`](https://crates.io/crates/imxrt-ral) to access the FlexIO registers.

I've no strong preference for a FlexIO driver design; if the FlexIO driver is dedicated for QSPI, that's good with me. Nevertheless, I think it could be cool to model the timers and shifters so that we could more easily build standard / custom peripherals. I have no idea if this is feasible though.

> I will be writing a SDCard library [...]

Awesome. See #119, which we can use to track any HAL developments.

mciantyre, 2022-09-28:
We've talked in other issues, and I want to double-check my understanding of this issue.

We're discussing a driver for the Flex**IO** peripheral (chapter 50 in the 1060 reference manual). We're _not_ discussing the Flex**SPI** controller (chapter 27 in the 1060 reference manual). The goal is to produce a FlexIO driver capable of QuadSPI. Is that the gist?

Sandvoxel, 2022-09-28:
That was a mistake on my part i meant FlexSpi in chapter 27.

mciantyre, 2022-09-28:
See also mciantyre/teensy4-rs#100, which asks about EEPROM emulation using FlexSPI.

tacertain, 2026-03-11:
I added support for PSRAM on the teensy4. I submitted #196 to add the clock locator. I put most of my changes in teensy4-rs. I don't know which if any really belong in imxrt-hal. I will submit those changes once these are accepted and/or I get feedback that some of these changes belong in here. Here is my commit to teensy4-rs that shows how to initialize the PSRAM: https://github.com/tacertain/teensy4-rs/commit/8e5e38a01254cb6b0cd972a57e55300fd2cd31df