imxrt-rs/imxrt-hal #123: Support for RT600

maxekman, 2023-01-28:
Hi,

This is more of a question; I would like to know if you would have an estimate for the amount of work needed to support the RT600 series? I'm not requesting it, just curious to know about your feeling for the effort needed.

I do understand that the additional DSP co-processor is completely different to the core CPU, and would probably better be implemented in it's own crate.

Thanks,
Max

teburd, 2023-01-28:
The rt600, specifically the rt685 has a different core (m33) and peripheral set than the rt1xxx series. The xtensa hifi4 core is an even more interesting task as there’s a forked rust compiler required to even attempt this from espressif and xtensa like its name suggests is a highly configurable core so it’d likely require some patches to support it well.

I think the m33 core would be mostly feasible. Get a ral output and merged with the existing ones were sensible. There’s a lot of unique peripherals for this part though, the powerquad dsp, the many many flexcomms, two dmacs, yeah it’s quite the chip! The evk board has a lot of nice audio interfaces already setup as well and those would be great to tinker with. I’ve gotten audio from the on board codec with C, the nxp samples, and zephyr. In rust this would be quite the achievement.

It would be an amazing experience to program this in rust though! At the moment even Zephyr doesn’t support it that well beyond the m33 core. Supporting this part fully would be a boon and open some doors, the hifi cores are used in many many places that you may not realize. Like intel uses them in many parts as can seen in the SoF project.

maxekman, 2023-01-29:
Thanks for the great summary! I did not realize that the Xtensa core would require a compiler fork, but I understand it now that you point it out. I also didn’t know that the Espressif chips used similar cores!

As you mention, I had that exact dev board in mind for an audio project. Sounds like quiet an undertaking to get it all working in Rust. I’ll let you know if I dive in, but most likely not.

maxekman, 2023-01-29:
Found the forked compiler, adding it here for future reference: https://github.com/esp-rs/rust

teburd, 2023-01-29:
It still might be worth getting the arm core supported even if the xtensa isn’t initially. They would require separate images anyways.