imxrt-rs/imxrt-hal #138: SPI Frame Size Limitation

Finomnis, 2023-06-07:
What's the reasoning for [the SPI frame size limit](https://docs.rs/imxrt-hal/latest/imxrt_hal/lpspi/enum.LpspiError.html#variant.FrameSize)?

> The frame size, in bits, must be between 8 bits and 4095 bits.

I haven't seen anything related to this in the `IMXRT1060XRM.pdf`; in fact, it seems to support 32 bit frame sizes.

What's the limitation, and if it's chip specific, could there be a solution that enables the other chips to support larger frame sizes?

My usecase: NeoPixel protocol emulation through SPI, which requires one contiguous stream of data.

mciantyre, 2023-06-07:
[image]

The frame size field, `FRAMESZ`, of the transmit command register (`TCR`) is 12 bits wide. 2<sup>12</sup> - 1 = 4095 was thought to be the upper bound. But looks like I have an off-by-one error in that docstring; the peripheral treats the field as `FRAMESZ + 1` bits. So the upper bound is 4096. (The implementation [already handles the off-by-one](https://github.com/imxrt-rs/imxrt-hal/blob/29df2d21ad03130d859c29dd97e87cf26c9a8abb/src/common/lpspi.rs#L282).)

The lower bound of 8 bits is described in the `FRAMESZ` field documentation, on the next page. From my study, the LPSPI peripheral is the same across all considered i.MX RT chips, so these bounds are not chip specific.

What implies that there's 2<sup>32</sup> bits per frame?

---

A continuous transfer might work for you. Continuous transfers keep chip select asserted at the end of frame, allowing us to transfer another frame for the same transaction. The larger the frame, the less frequently we need to re-submit a continuous transfer command to `TCR`.

There's more information in the LPSPI chapter of the reference manual. The lower-level `Lpspi` API [should support continuous transfers](https://docs.rs/imxrt-hal/latest/imxrt_hal/lpspi/struct.Transaction.html#continuous-transactions). The `embedded-hal` implementation uses continuous transfers for sending / receiving buffers of 8/16/32 bit words, so you can see how they're used by studying the `lpspi.rs` module.

Finomnis, 2023-06-07:
I apologize. I read

> If the frame size is larger than 32 bits, then the frame is divided into multiple words of 32 bits; each
word is loaded from the transmit FIFO and stored in the receive FIFO separately.

And read "addresses larger than 32bits" instead of "messages larger than 32bits".

I'm sorry I wasted your time on this one :)

Finomnis, 2023-06-07:
@mciantyre Quick quesition: Do you have any kind of FlexSPI support already? I mean, the FIFO mode that supports raw data transfer. Because FlexSPI does, to my knowledge, support way longer contiguous transfers.

closed 2023-06-07 by Finomnis

mciantyre, 2023-06-08:
You're all good! These kinds of questions give me the chance to review this work. Keep them coming.

Sorry, we've got nothing for FlexSPI. Keep up with #120 for developments. I haven't dug into the FlexSPI peripheral, and I haven't researched anyone using it for normal SPI. The rest of this is just riffing.

From the looks of the FlexSPI's `IPCR1` register, a single command could send 2<sup>15</sup> _bytes_. The internal IP TX FIFO is only 128 bytes large, but this could be saturated with a DMA channel. I'm not sure what the LUT sequence would look like to mimic normal SPI; maybe just a WRITE_SDR / READ_SDR instruction, followed by STOPs.