imxrt-rs/imxrt-hal #142: Road to `embedded-hal 1.0`

Finomnis, 2023-11-05:
This is a tracking issue of our progress and tasks to integrate the upcoming `1.0` version of `embedded-hal`.

TODO: Link the migration guide here

Tasks:
- [ ] Add a task here.

mciantyre, 2023-11-25:
No preference for or against breaking changes. Folks should feel free to explore new driver designs if they're warranted for embedded-hal 1.0. I'd still like to maintain some kind of embedded-hal 0.2 support. We could implement both traits with our drivers, or we recommend something like [embedded-hal-compat](https://docs.rs/embedded-hal-compat/latest/embedded_hal_compat/).

It might also be nice to support the async embedded-hal traits. My past attempts with async embedded i.MX RT drivers taught me what _not_ to do, so I'm looking for help here. I hope to see approaches that are decoupled from any async runtime, though I'm not sure if that's possible or a good idea.

Finomnis, 2023-11-25:
My plan is currently to implement purely the `async` version and convert them to blocking with the `cassette` crate. I do have quite some experience with async (I'm the owner of `tokio-graceful-shutdown` and rewrote tokio's `CancellationToken`), and I'm motivated to answer questions if someone needs help. :)

Finomnis, 2023-11-25:
I do agree with the plan to decouple from a specific runtime, and I don't think it's hard. So far for my lpspi rework I don't have anything that is specific to rtic or similar. E-h 1.0 so far seems will thought through.

Finomnis, 2023-12-21:
@mciantyre For timer adapters, can't we use [`rtic-monotonics`](https://docs.rs/rtic-monotonics/latest/rtic_monotonics/imxrt/index.html)? While having `rtic` in its name, it's actually runtime agnostic.

Finomnis, 2024-05-21:
@mciantyre Would you mind an intermediate release, now that implemented GPIO and UART for e-h 1.0? I need that for a project of mine and would like to official release it, but it depends on the github version of `imxrt-hal` because of this.

prutschman, 2025-04-20:
Apologies if this is the wrong forum. I see that the roadmap checkbox indicates timer work is done, and that adapters are deferred to rtic-monotonic. Blocking appears to impl eh02::blocking::delay::{DelayMs,DelayUs}, but not eh1::delay::DelayNs. Is this an oversight, or am I missing something about the new way of doing things? I didn't see anything about DelayNs in rtic-monotonic.

Finomnis, 2025-04-21:
@prutschman I'm unsure at you mean with "blocking appears to impl".

Finomnis, 2025-04-21:
@prutschman be aware that `rtic-monotonic` and `rtic-monotonics` are two different crates. You are probably looking at the wrong one.

This is the one you should be using: https://docs.rs/rtic-monotonics/latest/rtic_monotonics/imxrt/index.html

prutschman, 2025-04-21:
Sorry, that was unclear shorthand. I mean that there are implementations of the embedded-hal 0.2.x "blocking::DelayMs" and "blocking::DelayUs" traits for the imxrt-hal "Blocking" struct, but there didn't appear to be a corresponding implementation of the embedded-hal 1.0 "DelayNs" trait, their successor.

Because of the mention of rtic-monotincs (thank you for the clarification) I thought to look there for the implementation, but it appears to deal with other concerns.

Finomnis, 2025-04-21:
The entire `imxrt-hal::timer::BlockingGpt` isn't really used any more, the new implementation you should use is in https://docs.rs/rtic-monotonics/latest/rtic_monotonics/imxrt/.

It says `rtic` in its name, but isn't actually connected to the rtos. It's a general purpose timer for various chips; and as it exists and I am kind of is author there isn't much of a point duplicating the timer code to this crate.

Finomnis, 2025-04-21:
Of course if it's useful to have an eh1 impl of the `Blocking` struct it's possible to add one. Feel free to create a PR, it probably shouldnt be too hard to achieve, I think.

Finomnis, 2025-04-21:
Does that make sense? If not, where are you stuck?

prutschman, 2025-04-21:
I'm willing to take a look at a making a PR for a DelayNs implementation, unless blocking delays are really an ecosystem dead-end. I see the advantage to using Monotonic in a project that's already gone full-async. In my case I was looking at updating some existing work that used DelayMs to embedded-hal 1.0. (It's an i2c device driver that currently takes a &mut DelayMs to sequence some initialization operations).

Finomnis, 2025-04-21:
Be aware that the Monotonics *do* implement eh1 DelayNs, they are not just for async. They have blocking calls as well.

prutschman, 2025-04-22:
So it does! I missed it because I didn't see the implemented traits in the docs, which makes sense if it's in a macro expression. I will definitely keep that in mind. Unfortunately there doesn't seem to be a Monotonic implementation for the PITs, which is what I'm currently using. But I will see if I can rearrange things to free up the other GPT in my application. If not, I'll revisit making a pull request for DelayNs. Thank you for your assistance.