imxrt-rs/imxrt-hal #144: LPSPI3 Init Hanging

N8BWert, 2023-11-11:
I'm currently using the Teensy 4.1 for a project and need to utilize bother the SPI0 (LPSPI4) and SPI1 (LPSPI3) instances.  I have not problem with utilizing the LPSPI4 instance, however whenever the LPSPI3 instance is instantiated (i.e. caling hal::lpspi::Lpspi::new() with the LPSPI3 instance) the program hangs.

An example program that will cause the on-board led to blink 9 times (indicating the program stalled during initialization) is:
```rust
#![no_std]
#![no_main]

use bsp::board::{self, LPSPI_FREQUENCY};
use teensy4_bsp as bsp;
use teensy4_panic as _;

use bsp::hal as hal;
use bsp::ral::lpspi::LPSPI3;

#[bsp::rt::entry]
fn main() -> ! {
    let board::Resources {
        pins,
        ..
    } = board::t41(board::instances());

    let spi_pins = hal::lpspi::Pins {
        pcs0: pins.p38,
        sck: pins.p27,
        sdo: pins.p26,
        sdi: pins.p39,
    };
    let spi3 = unsafe { LPSPI3::instance() };
    let mut radio_spi = hal::lpspi::Lpspi::new(spi3, spi_pins);

    radio_spi.disabled(|spi| {
        spi.set_clock_hz(LPSPI_FREQUENCY, 1_000_000);
    });

    loop {

    }
}
``` 

I'm not sure what the issue could be but any help would be much appreciated.  Thank you.

mciantyre, 2023-11-11:
I haven't tested on hardware, but I'm pretty sure this is a teensy4-bsp issue. The BSP isn't enabling the LPSPI**3** clock gate. Within [this collection](https://github.com/mciantyre/teensy4-rs/blob/833366b4e0de92b9c70cc664ed51c43833ab897a/src/clock_power.rs#L191) in the teensy4-bsp package, try adding a call to

```rust
clock_gate::lpspi::<3>(),
```

This should automatically turn on the clock gate. If that works, your PR with the fix would help me expedite a fix.

If this is your first time testing custom patches in your dependencies, the [overriding dependencies](https://doc.rust-lang.org/cargo/reference/overriding-dependencies.html) guide might be helpful.

N8BWert, 2023-11-11:
That was it.  Thank you so much.  I've created the PR [here](https://github.com/mciantyre/teensy4-rs/pull/149)

closed 2023-11-11 by N8BWert