imxrt-rs/imxrt-hal #150: Initializing the SRTC crashes the board

therealbstern, 2024-01-26:
This is on a Teensy 4.0 using `teensy4-rs`.

When I initialize the SRTC, the code stops running.

A minimal example is at https://github.com/therealbstern/srtc-error

This worked with `teensy4-rs` 0.3 and `imxrt-ral` 0.4 but doesn't work with `teensy4-rs` 0.4 and `imxrt-ral` 0.5.

mciantyre, 2024-01-26:
This issue doesn't reproduce using the `hal_srtc` example in this project, so it looks like the issue is in the BSP. Could you give mciantyre/teensy4-rs#154 a try? I think this patch directive in a Cargo.toml file should work:

```toml
[patch.crates-io.teensy4-bsp]
git = "https://github.com/mciantyre/teensy4-rs"
branch = "snvs-hp-cg"

```

therealbstern, 2024-01-26:
I will give it a try and report back.  Thank you for your help!

therealbstern, 2024-01-27:
That fixed my test case, but my longer program which uses more accessories doesn't run any more.  I'll try to figure out where it's crashing and make another example.

mciantyre, 2024-01-27:
Thanks for testing. I released teensy4-bsp 0.4.7 with this fix. I deleted the `snvs-hp-cg` branch in the teensy4-rs repo, so that patch directive might not work anymore.

If you notice any other problems, open issues here or in the teensy4-rs project.

closed 2024-01-27 by mciantyre

therealbstern, 2024-01-27:
FYI, my more complicated program crashes any time I call a macro, but that's clearly something else.  Thanks again!