imxrt-rs/imxrt-hal #184: Impossible to get I2C running

EzraWolf, 2025-01-02:
For over a month I have been trying to get a simple I2C program running, and for over a month I have received exactly one error: `BUS_BUSY | ARBITRATION_LOST` which is likely due to the I2C controller flag `TRANSMIT_DATA` being high.

I don't know enough about the I2C protocol to understand it, and I do not have a digital analyzer or oscilloscope to measure the lines. Hardware looks fine, I am using a Teensy 4.1 and it is connected to an SSD1306 OLED will pullups.

Here is what I know.

TRANSMIT_DATA flag is stuck high from initialization one way or another, have never seen it low in any case no matter what condition at all. Cannot be cleared with any of the following:

- Using reset_controller()
- Using clear_fifo()
- Using clear_controller_status(ControllerStatus::all())
- Trying thousands of reset attempts in a while loop out of desperation

When attempting I2C write:
- System hangs (perpetual LED of sisyphus stops blinking)
- Need to remove power from the OLED before anything runs again
- Afterwards the USB logger shows ARBITRATION_LOST and BUS_BUSY errors from the I2C write_byte function
- FIFO shows txcount of 2 after failed write *sometimes* ??

Hardware Setup
- Teensy 4.1
- SSD1306 OLED Display
- I2C Address: 0x3C
- Using LPI2C3 peripheral on pin 16 for SCL and pin 17 for SDA, LPI2C1 has the exact same issues
- Both pins configured with internal 100k pull-ups

Attempted Solutions
- Every combination of resetting things in different orders (e.g., clear FIFO then I2C controller, etc.)
- Tried to clear status flags in different orders as well
- Tried 100KHz and 400KHz each numerous times
- Explicit pin configuration before I2C initialization
- Multiple reset attempts during initialization
- Read through most of the embedded-hal and imxrt-hal source code
- Read each of their respective doc books front and back and could not find any issues or implementations
- Implementing 2 I2C examples listed in the referenced code, neither worked.

I am about to give up on this insultingly simple problem. Please anybody help, if anything to see a different error. I have spent the better part of 60+ hours on this.

Referenced code:
https://github.com/RoboJackets/robocup-rustware/blob/2fab2e23d003bdc0cc6d33cc6efbb2a6bbc87f69/control/examples/scan_i2c.rs
https://github.com/SharpCoder/teensycore/blob/8bd5535aacf34af972a07a849f02c8160741a96f/src/i2c.rs
every single example: https://github.com/mciantyre/teensy4-rs/tree/master/examples
uses an old version of embedded-hal i could not determine to get working: https://github.com/imxrt-rs/imxrt-hal/blob/main/examples/hal_i2c_mpu9250.rs

The entirety of my code is listed below.

```rust

#![no_std]
#![no_main]
#![deny(unsafe_code)]

use teensy4_panic as _;

mod drivers;

#[rtic::app(device = teensy4_bsp, peripherals = true, dispatchers = [KPP])]
mod app {

    // Teensy 4.0 and 4.1 board imports
    // - Teensy 4.1 is `teensy_board::t41`
    // - Teensy 4.0 is `teensy_board::t40`
    use teensy4_bsp as bsp; // Target System on a Module (SoM)

    // IMXRT HAL imports
    use imxrt_log as logging;

    // RTIC imports
    // TODO: "*" is for `.millis()`, need to change
    use rtic_monotonics::systick::{Systick, *};
[... 309 lines cut ...]
features = ["cortex-m-systick"]

###########################
# Build-time dependencies #
###########################
[build-dependencies]
toml = "0.8.19"

# Set to 0 to optimize build times, no optimizations
# Set to 3 for all optimizations
# Set to "z" to optimize build size w/o loop vectorization
[profile.release.build-override]
# opt-level = 0 # No optimizations
# opt-level = 1 # Few optimizations
# opt-level = 2 # Some optimizations
# opt-level = 3 # All optimizations (slowest build time)
# opt-level = "s" # Smallest binary size
opt-level = "z" # Optimize for build size w/o loop vectorization

# [target.<architecture-triple>]
# runner = 'probe-rs run --chip <chip-name>'

```

```

EzraWolf, 2025-01-08:
There is no solution to this?

teburd, 2025-01-09:
I would *highly* recommend getting an inexpensive logic analyzer to view what is going on here