imxrt-rs/imxrt-hal #189: Teensy 4.1 support?

libesz, 2025-07-07:
Hi! I just started to port my project from stm32 to teensy 4.1 and I ran into a 128k code size limit, using the `board/teensy4` feature.
Is it so that the linker script is prepared for teensy 4.0, while also compatible with 4.1 with constraints? Would there be an easy/existing way to tweak the linker script for teensy 4.1, or it would mean more work?

Thanks!

mciantyre, 2025-07-07:
Sounds like you're building your project in tree. Here's where the runtime configuration happens:

https://github.com/imxrt-rs/imxrt-hal/blob/6a3b53d7ef4abee86375394f367d1cf91ef94626/board/build.rs#L46-L53

The [imxrt-rt](https://docs.rs/imxrt-rt/0.1.7/imxrt_rt/) package defines the linker script and how much RAM is available for instructions. See its documentation for more information.

Make sure you're building with optimizations. Then, try increasing the instruction RAM. When it comes to instruction RAM, I wouldn't think that the Teensy 4.0 vs 4.1 distinction matters.

libesz, 2025-07-07:
> Sounds like you're building your project in tree.

Not sure about this.

I am atm writing an SGTL5000 driver crate. In the examples I attempted to pull imxrt-hal and /board, and a big audio library dependency (which my main project would use in the future), so I hit the 128k limit with .text. NXP and PJRC documentation suggests that there should be at least 512k for itcm on the 1062 chip. My guess was it is maybe something related to the teensy 4.0 and 4.1 differences.

libesz, 2025-07-08:
Without fully understanding what I am doing 😄  [this commit](https://github.com/libesz/imxrt-rt/commit/670af23badbf1adb81099894350fef78218575f7) made this change in the resulted linker file:

from
```
MEMORY {
FLASH (RX) : ORIGIN = 0x60000000, LENGTH = 0x1F0000
ITCM (RWX) : ORIGIN = 0x0, LENGTH = 0x20000
DTCM (RWX) : ORIGIN = 0x20000000, LENGTH = 0x20000
OCRAM (RWX) : ORIGIN = 0x20200000, LENGTH = 0xC0000
}
```

to
```
MEMORY {
FLASH (RX) : ORIGIN = 0x60000000, LENGTH = 0x1F0000
ITCM (RWX) : ORIGIN = 0x0, LENGTH = 0x40000
DTCM (RWX) : ORIGIN = 0x20000000, LENGTH = 0x20000
OCRAM (RWX) : ORIGIN = 0x20200000, LENGTH = 0xA0000
}
```

This allowed me to compile my 128k+ binary, yay!

I wonder if it would be feasible to support the same layout as in case of arduino: https://github.com/PaulStoffregen/cores/blob/master/teensy4/imxrt1062_t41.ld#L3-L7

teburd, 2025-07-08:
> SGTL5000

Super cool, do check out the device-driver crate for generating regmap accessors, it looks neat and I wish I had something like it in C for what I do